oliexdev/openScale #1084: Openscale Sync crashes after adding mqtt server info

tobox, 2024-11-07:
**Describe the bug**

I bought and installed openscale sync and tried to enter mqtt credentials. It did not work with TLS, so I tried without encryption. Then openscale crashed repeatedly.

**To Reproduce**
Steps to reproduce the behavior:

Using mosquitto and a local hostname (mosquitto.fritz.box in my case), added username and password, and the crash occured.
Openscale sync was the current version from today in play store.

**Expected behavior**
Should not crash.

**Additional context**
Build version: 0.3.5 
Build date: 1981-01-01 01:01:02 
Current date: 2024-11-07 09:09:46 
Device: Google Pixel 6 
OS version: Android 15 (SDK 35) 
 
Stack trace:  
java.lang.SecurityException: com.health.openscale.sync: One of RECEIVER_EXPORTED or RECEIVER_NOT_EXPORTED should be specified when a receiver isn't being registered exclusively for system broadcasts
	at android.os.Parcel.createExceptionOrNull(Parcel.java:3242)
	at android.os.Parcel.createException(Parcel.java:3226)
	at android.os.Parcel.readException(Parcel.java:3209)
	at android.os.Parcel.readException(Parcel.java:3151)
	at android.app.IActivityManager$Stub$Proxy.registerReceiverWithFeature(IActivityManager.java:6092)
	at android.app.ContextImpl.registerReceiverInternal(ContextImpl.java:1881)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1821)
	at android.app.ContextImpl.registerReceiver(ContextImpl.java:1809)
	at android.content.ContextWrapper.registerReceiver(ContextWrapper.java:757)
	at i4.a.start(:2)
	at t5.b.q(:9)
	at t5.d.run(Unknown Source:91)
	at java.lang.Thread.run(Thread.java:1012)
Caused by: android.os.RemoteException: Remote stack trace:
	at com.android.server.am.ActivityManagerService.registerReceiverWithFeatureTraced(ActivityManagerService.java:14828)
	at com.android.server.am.ActivityManagerService.registerReceiverWithFeature(ActivityManagerService.java:14630)
	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:2757)
	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:2812)
	at android.os.Binder.execTransactInternal(Binder.java:1500)

newmanew1, 2024-12-31:
I've a very similar issue - it seems to be android version dependant:
https://stackoverflow.com/questions/77235063/one-of-receiver-exported-or-receiver-not-exported-should-be-specified-when-a-rec

Foguet, 2025-02-17:
I have the same problem, is there any solution? (it is to return the app because without google fit, or mqtt, it is useless for me right now) :(

oliexdev, 2025-05-10:
This problem might be resolved in the upcoming beta version (0.4) of openScale sync, which features a complete rewrite of the application and includes improvements to the synchronization logic.

We are currently running a closed beta test for this new version. If you'd like to help test and see if this issue is fixed, please consider joining the beta program. You can find more details and instructions on how to join in our pinned beta testing invitation issue here: https://github.com/oliexdev/openScale/issues/1124

closed 2025-05-10 by oliexdev