oliexdev/openScale #1094: Regarding Reverse Engineering

PrasannaVenkadesh, 2024-12-18:
Following the reverse engineering guide, I am trying to reverse engineer the Bluetooth packets of a Bluetooth weighing scale. However, I am only seeing L2CAP packets and not any ATT protocol packets. Does that mean, the data is encrypted?

Weighing Scale: [Healthsense S1](https://healthsense.in/products/healthsense-s1-luxury-smart-bluetooth-body-weighing-scale?variant=40693780644023)
Vendor App: [Fitdays](https://play.google.com/store/apps/details?id=cn.fitdays.fitdays)

[image]

PrasannaVenkadesh, 2024-12-18:
Just realized that the device I am dealing with is also from ICOMON and I also see that initial support for these devices are added by @Duncaen

Hi @Duncaen if you could share how to decode the packets, I can contribute to the reverse engg efforts!

PrasannaVenkadesh, 2024-12-18:
I did some research and found out that these devices does not implement any GATT services, instead transmits data or rather broadcasts data using BLE Advertising packets. I am pretty much new to Bluetooth in general, but finds it fascinating and interesting.

Any handy resources and tools that helps in reverse engineering Broadcast devices would be really helpful.

Duncaen, 2024-12-18:
I don't really have any specific suggestions or insight into this specific scale because the same app supports like 100 different variants. But chances are if its also a broadcast scale you can use the same algorithm. https://github.com/oliexdev/openScale/issues/593#issuecomment-2067806719

I basically used jadx to decompile the app and then some parts of the code "decoding" happens in a native library (c++) that I've reverse engineered with ghidra.

PrasannaVenkadesh, 2024-12-19:
Awesome. Thanks @Duncaen

Whatever I was looking for has already been discussed in that thread and that too 4 years ago!

I guess I will close this issue.

closed 2024-12-19 by PrasannaVenkadesh