oliexdev/openScale #1097: Add feature to vibrate or make a sound when logged data

adeekshith, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
No indication the data is logged when the user is on the scale without looking at the app 

**Describe the solution you'd like**
A notification or haptic feedback to let the user know data is logged

BenBarat, 2024-12-30:
A annoyance of mine for ages. Quite important for those with eyesight issues and other disabilities. **Accessibility compliance** with regulations in different countries, and future versions of Android.

There should be a toggle choice in the settings menu (call it '_Haptic feedback_') to select sound, vibrate, or both, when a data point has been successfully recorded (ie: at the end of the loop). This preference should be stored in the local configuration, by user, as not everybody would want this. What do you think - initially turned off or on as default setting?

As another further option, choice of sound would be good: loud boing, soft peep, a rising whistle, the word OK/Yes/teenage grunt in local language (in adult male and female and child male or female), one the user has recorded themselves (or maybe their favorite performer), or one of the ones available in the standard Android repertoire to save on app download size.

You could have your life couch enthusiastically give you a mini pep talk, scream at you, sigh, or tut-tut, depending on what change has happened between your last reading and the current one. Up, down, large or small % variation would be choices you could configure.

You could also haptically prompt when asked to initially step on the scale. Soothing words of guile and enticement.

You could charge for this as an add-on for **sound themes**, for different performers, pop stars, movie stars, politicians, motivational speakers, influencers, festive seasons, or just familiar family members for those with dementia or panic syndrome that need a friendly sound. Daleks/Dr Who, Star Wars, Taylor Swift, favourite BTS performer, your mom, your favorite sports team song, your evil step-sister for Halloween - you get the idea.

A theme generator/previewer could then be bundled with the package, or hints for AI packages to be guided by, to artificially generate suitable sounds.

Sorry to jump in. A lot of extra coding to get something flashy.

For starters, just adding a simple boing  and a gentle shake would be great.

BenBarat, 2024-12-30:
As a start, just add a notification for measurement value and unit of measurement and let the Android notification subsystem take care of it for turning the sound and vibration off/on.

Sound themes can follow later.