oliexdev/openScale #1099: App error when trying to connect to Etekcity ESF24

Oaklight, 2025-01-03:
**Describe the bug**
App error pops up as soon as connecting to the Bluetooth scale.

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 2.5.3

**Expected behavior**
Not crash the app.

**Debug log**
```
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-01-03 05:32:30 
Device: HUAWEI JAD-AL50 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:966)
	at android.os.Handler.dispatchMessage(Handler.java:110)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:293)
	at android.app.ActivityThread.loopProcess(ActivityThread.java:9986)
	at android.app.ActivityThread.main(ActivityThread.java:9975)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:586)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1240)
```

Dwarf1er, 2025-01-03:
I also have an Etekcity ESF24 and got a crash when trying to connect

**Describe the bug**
App crashed when trying to connect to the scale

**Expected behaviour**
Connect to the scale without crashing the app

**Debug log**
```
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-01-03 15:43:32 
Device: Google Pixel 7a 
OS version: Android 15 (SDK 35) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:991)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8826)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ExecInit.main(ExecInit.java:50)
	at com.android.internal.os.RuntimeInit.nativeFinishInit(Native Method)
	at com.android.internal.os.RuntimeInit.main(RuntimeInit.java:369)
```

oliexdev, 2025-03-03:
somebody has to test the following PR --> https://github.com/oliexdev/openScale/pull/1083

Oaklight, 2025-03-03:
I'm happy to test it, not sure if this pr feature already available in today's dev-build. But I have tried and got this 
```
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-03-03 04:58:50 
Device: HUAWEI JAD-AL50 
OS version: Android 12 (SDK 31) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Objects.java:244)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:966)
	at android.os.Handler.dispatchMessage(Handler.java:110)
	at android.os.Looper.loopOnce(Looper.java:205)
	at android.os.Looper.loop(Looper.java:293)
	at android.app.ActivityThread.loopProcess(ActivityThread.java:9986)
	at android.app.ActivityThread.main(ActivityThread.java:9975)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:586)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1240)
```

oliexdev, 2025-03-03:
@Oaklight that was the wrong dev build please try out the following --> https://github.com/oliexdev/openScale/releases/tag/dev-build

Oaklight, 2025-03-03:
I just downloaded the dev-build you pointed to. It was able to connect the scale, but nothing happened after seeing the block notice showing xxxx is connected. no read from scale showing up on the app, nor manual adding entry picking up correct current weight. Any way to send you the log?

Dwarf1er, 2025-03-13:
Hi! I tried the dev build you linked and had the same result as @Oaklight , it successfully connects but there's no read showing up in the app.

Btw @Oaklight the way to get the log file is detailed in the Wiki [here](https://github.com/oliexdev/openScale/wiki/Frequently-Asked-Questions-%28FAQ%29#how-to-get-the-openscale-debug-log-file)!

Here's my log file:
```md
2025-03-13 17:37:52.801 Debug [2] AboutPreferences: Debug log enabled, openScale v2.5.2 (64), SDK 35, Google Pixel 7a
2025-03-13 17:37:52.805 Debug [2] AboutPreferences: Selected user id(1) name(Test) birthday(Fri Dec 31 19:00:00 EST 1999) age(25) body height(160.02) scale unit(lb) gender(male) initial weight(70.31) goal enabled(false) goal weight(-1.00) goal date(Thu Mar 13 17:34:09 EDT 2025) measure unt(in) activity level(0) assisted weighing(false)
2025-03-13 17:37:57.460 Debug [2] OpenScale: Disconnecting from bluetooth device
2025-03-13 17:37:57.464 Debug [2] BluetoothCommunication: Bluetooth disconnect
2025-03-13 17:37:57.465 Info [2] BluetoothCentralManager: no scan to stop because no scan is running
2025-03-13 17:37:57.467 Error [2] BluetoothCentralManager: cannot cancel connection to unknown peripheral D8:0B:CB:2A:29:22
2025-03-13 17:37:58.209 Debug [2] MainActivity: Main Activity Bluetooth permission check
2025-03-13 17:37:58.214 Debug [2] MainActivity: SDK >= 31 request for Bluetooth Scan and Bluetooth connect permissions
2025-03-13 17:37:58.219 Debug [2] MainActivity: Permission already granted: android.permission.BLUETOOTH_SCAN
2025-03-13 17:37:58.223 Debug [2] MainActivity: Permission already granted: android.permission.BLUETOOTH_CONNECT
2025-03-13 17:37:58.233 Debug [2] OpenScale: Trying to connect to bluetooth device [D8:0B:CB:2A:29:22] (QN-Scale)
2025-03-13 17:37:58.241 Debug [2] BluetoothCommunication: Do LE scan before connecting to device
2025-03-13 17:37:58.256 Info [2] BluetoothCentralManager: scan started
2025-03-13 17:37:58.259 Debug [2] BluetoothCommunication: Stop machine state
2025-03-13 17:38:01.334 Debug [2] BluetoothCommunication: Found peripheral 'QN-Scale'
2025-03-13 17:38:01.349 Info [2] BluetoothCentralManager: scan stopped
2025-03-13 17:38:02.359 Debug [2] BluetoothCommunication: Try to connect to BLE device D8:0B:CB:2A:29:22
2025-03-13 17:38:02.473 Info [2] BluetoothPeripheral: connect to 'QN-Scale' (D8:0B:CB:2A:29:22) using transport LE
2025-03-13 17:38:02.507 Info [2] BluetoothPeripheral: peripheral 'D8:0B:CB:2A:29:22' is connecting
2025-03-13 17:38:02.735 Info [570] BluetoothPeripheral: connected to 'QN-Scale' (NONE) in 0.2s
2025-03-13 17:38:02.744 Debug [2] BluetoothPeripheral: discovering services of 'QN-Scale' with delay of 0 ms
2025-03-13 17:38:03.020 Debug [570] BluetoothPeripheral: connection parameters: interval=7.5ms latency=0 timeout=5s
2025-03-13 17:38:03.322 Info [570] BluetoothPeripheral: discovered 5 services for 'QN-Scale'
2025-03-13 17:38:03.347 Debug [2] BluetoothCommunication: connected to 'QN-Scale'
2025-03-13 17:38:03.360 Debug [2] BluetoothCommunication: Successful Bluetooth services discovered
2025-03-13 17:38:03.370 Debug [2] BluetoothCommunication: Resume machine state
2025-03-13 17:38:03.378 Debug [2] BluetoothCommunication: Step Nr 0
2025-03-13 17:38:03.388 Debug [2] BluetoothCommunication: Invoke write bytes [02 6B C6 65 2F] on 0xffe4
2025-03-13 17:38:03.395 Debug [2] BluetoothCommunication: Step Nr 1
2025-03-13 17:38:03.400 Debug [2] BluetoothCommunication: Invoke set notification on 0xfff1
2025-03-13 17:38:03.404 Debug [570] BluetoothPeripheral: connection parameters: interval=30.0ms latency=0 timeout=5s
2025-03-13 17:38:03.413 Debug [2] BluetoothCommunication: Stop machine state
2025-03-13 17:38:03.427 Debug [2] MainActivity: Bluetooth connection successful established
2025-03-13 17:38:03.492 Debug [2] BluetoothCommunication: SUCCESS: Notify set for 0000fff1-0000-1000-8000-00805f9b34fb
2025-03-13 17:38:03.494 Debug [2] BluetoothCommunication: Resume machine state
2025-03-13 17:38:03.495 Debug [2] BluetoothCommunication: Step Nr 2
2025-03-13 17:38:03.498 Debug [2] BluetoothCommunication: Invoke write bytes [13 09 15 02 10 00 00 00 43] on 0xfff2
2025-03-13 17:38:03.500 Debug [2] BluetoothCommunication: Step Nr 3
2025-03-13 17:38:03.503 Debug [2] BluetoothCommunication: Invoke write bytes [02 6B C6 65 2F] on 0xfff2
2025-03-13 17:38:03.505 Debug [2] BluetoothCommunication: Step Nr 4
2025-03-13 17:38:03.506 Debug [2] BluetoothCommunication: Step Nr 5
2025-03-13 17:38:03.508 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2025-03-13 17:38:03.511 Debug [2] BluetoothPeripheral: writing <130915021000000043> to characteristic <0000fff2-0000-1000-8000-00805f9b34fb>
2025-03-13 17:38:03.515 Debug [2] MainActivity: Bluetooth scale message: Please step barefoot on the scale
2025-03-13 17:38:03.553 Debug [2] BluetoothCommunication: SUCCESS: Writing <13 09 15 02 10 00 00 00 43> to <0000fff2-0000-1000-8000-00805f9b34fb>
2025-03-13 17:38:03.555 Debug [2] BluetoothCommunication: Step Nr 5
2025-03-13 17:38:03.557 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2025-03-13 17:38:03.560 Debug [2] BluetoothPeripheral: writing <026bc6652f> to characteristic <0000fff2-0000-1000-8000-00805f9b34fb>
2025-03-13 17:38:03.611 Debug [2] BluetoothCommunication: SUCCESS: Writing <02 6B C6 65 2F> to <0000fff2-0000-1000-8000-00805f9b34fb>
2025-03-13 17:38:03.613 Debug [2] BluetoothCommunication: Step Nr 5
2025-03-13 17:38:03.615 Debug [2] BluetoothCommunication: Invoke delayed disconnect in 60s
2025-03-13 17:38:03.732 Debug [2] BluetoothQNScale: 14 0B 15 00 00 01 00 00 00 00 35 
2025-03-13 17:38:03.957 Debug [570] BluetoothPeripheral: connection parameters: interval=30.0ms latency=0 timeout=4s
2025-03-13 17:38:20.766 Debug [2] AboutPreferences: Debug log disabled

```

closed 2025-03-15 by oliexdev