oliexdev/openScale #1103: Scale support for Xiaomi Body Composition Scale S400

Sakalar, 2025-01-10:
Xiaomi Body Composition Scale S400

Is it possible to support this scale? This should be possible with the following link, right?

https://github.com/AlexxIT/XiaomiGateway3/issues/1388)

LostName1, 2025-06-11:
+1, is openscale still being developed?

gitauto-pro, 2025-07-03:
+1 for Xiaomi Body Composition Scale S400

Sakalar, 2025-08-26:
Mi-scale-exporter supports the Mi scale S400. Why not openscale?
https://github.com/lswiderski/mi-scale-exporter

RobertWojtowicz, 2025-08-26:
Hi, @Sakalar 

Check out my version, I figured it out my way, maybe it will help:
https://github.com/RobertWojtowicz/export2garmin/blob/master/manuals/S400_BLE.md
https://github.com/RobertWojtowicz/export2garmin

BR,
Robert