oliexdev/openScale #1105: Fitbit Aria Air support (unlikely?)

drinkcat, 2025-01-12:
Device: [Fitbit Aria Air](https://store.google.com/product/fitbit_aria_air)

That scale a bit flaky with the official app, so I was hoping to make it work with Openscale.

Sadly, it seems like the data is encrypted in some way, so, support seems unlikely (no idea where the key is, I also captured pairing process and couldn't see anything obvious).

I'm dropping this here to leave a trace behind of my experiments yesterday (so that others don't waste time, or at least know what they are getting into...), but if there's any simple experiment I can try, I'd be happy to.

A sample `Handle Value Indication` packet looks like this after decoding in wireshark (removed some not relevant data for privacy):
```
Frame 4277: 31 bytes on wire (248 bits), 31 bytes captured (248 bits)
    Encapsulation type: Bluetooth H4 with linux header (99)
    ...
    Frame Number: 4277
    Frame Length: 31 bytes (248 bits)
    Capture Length: 31 bytes (248 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    Point-to-Point Direction: Received (1)
    [Protocols in frame: bluetooth:hci_h4:bthci_acl:btl2cap:btatt]
Bluetooth
    [Source: TelinkSemico_XXX (a4:c1:38:XXX)]
    [Destination: XXX]
Bluetooth HCI H4
    [Direction: Rcvd (0x01)]
    HCI Packet Type: ACL Data (0x02)
Bluetooth HCI ACL Packet
    .... 0000 0100 0010 = Connection Handle: 0x042
    ..10 .... .... .... = PB Flag: First Automatically Flushable Packet (2)
    00.. .... .... .... = BC Flag: Point-To-Point (0)
    Data Total Length: 26
    Data
    [Connect in frame: 4054]
    [Disconnect in frame: 4343]
    [Source BD_ADDR: TelinkSemico_XXX (a4:c1:38:XXX)]
    [Source Device Name: Aria Air]
    [Source Role: Unknown (0)]
    [Destination BD_ADDR: XXX]
    [Destination Role: Unknown (0)]
    [Current Mode: Unknown (-1)]
Bluetooth L2CAP Protocol
    Length: 22
    CID: Attribute Protocol (0x0004)
Bluetooth Attribute Protocol
    Opcode: Handle Value Indication (0x1d)
        0... .... = Authentication Signature: False
        .0.. .... = Command: False
        ..01 1101 = Method: Handle Value Indication (0x1d)
    Handle: 0x0019 (Weight Scale: Unknown)
        [Service UUID: Weight Scale (0x181d)]
        [UUID: 67b0ab2c8323427ab1c970324bb5e228]
    Value: 9fbaa50ba5ba75ea3822d2f02305a1db3858f5
```

I was trying to make sense of the value, and... it looks totally scrambled.

These are all different weights:
```
0242201a00160004001d1900a8e870ec70e8d1a2c790bac096bd6fd2268fc8
0242201a00160004001d1900a12352015223478e9aa7d6f265c6957fdcc63a
0242201a00160004001d19009fbaa50ba5ba75ea3822d2f02305a1db3858f5
0242201a00160004001d1900d0a6081708a64d9b545a5502e8347ea173b81f
```

These are all the same weight (same as the first one above actually):
```
0241201a00160004001d190040f26ab96af2e4c91c2e426e9586d890df8edb
0241201a00160004001d1900eb1603b303162d5bbb386eda842a6089af6b9e
0241201a00160004001d19008c4becaaec4b972e157350756bd077864c49b9
0241201a00160004001d19005ec8b29bb2c890219832d283c908ff906d18cb
0241201a00160004001d19003c05019701050a14763693768510177e53fd6a
```

AshUK, 2025-04-15:
@drinkcat When setting up the scale after a factory reset, the BLE ATT handle 0x001d appears to have 16bytes written. Could be a key set up happening here ?

drinkcat, 2025-04-15:
@AshUK Yes I assume there is some key exchange happening here. Didn't dig further (for all we know, some interaction with fitbit servers could even be required -- I have no idea).

I also wasn't even sure if a custom implementation would solve the frequent BT issues with that scale.

Ended up going for a dumber solution (but hard to implement...) and OCRed the displayed numbers: https://github.com/drinkcat/SmarterScale . Works well enough for me.

ELY3M, 2025-12-07:
I guess I have to trash my scale :( and buy a new one that this app will work with   
I have Fitbit aria air scale.

orionshock, 2026-10-26:
I had AI do some reverse engineering on the scale, here are its findings:::

**BLE measurement protocol**

The Aria Air (FB203) uses AES-CCM to encrypt its weight measurements. Each indication is 19 bytes:

- Bytes 0–7: 8-byte nonce
- Bytes 8–15: 8-byte authentication tag
- Bytes 16–18: 3-byte ciphertext

The decrypted plaintext consists of a flags byte followed by a little-endian unsigned 16-bit weight value.

- Flags bit 0 set: `weight = raw * 0.01` (lbs)
- Flags bit 0 clear: `weight = raw * 0.005` (kg)

The measurement characteristic is:

`67b0ab2c-8323-427a-b1c9-70324bb5e228`

For AES-CCM implementations expecting ciphertext followed by the authentication tag:

```python
nonce = packet[0:8]
tag = packet[8:16]
ciphertext = packet[16:19]

plaintext = aesccm.decrypt(nonce, ciphertext + tag, b"")
```

The AES-CCM tag length is 8 bytes.

**Key provisioning**

This was the more interesting part.

The provisioning characteristic is:

`26f33a01-52a8-414b-99a2-1db75c99c032`

Reading it returns `01` when the scale is provisioned. After performing the physical factory reset (`clr`), it returned `00`.

Analysis of the Fitbit Android application indicated that provisioning consists of writing raw AES key bytes to this characteristic, without additional headers or authentication tokens in the write payload.

I tested this by generating a random 16-byte AES key locally and writing it to the reset scale using a normal GATT write with response.

The write succeeded, and the provisioning status changed to `01`.

More importantly, the scale subsequently produced measurements that could be authenticated and decrypted using that same key.

**Validation**

- 12 captured measurements successfully authenticated with AES-CCM.
- All 12 decoded values matched the scale's displayed readings.
- The scale continued using the same key after removing and reinstalling its batteries.
- No firmware modifications were necessary.
- No Fitbit API or account access was used during provisioning or measurement decoding.

This was tested on one Aria Air (FB203), so I can't say whether other firmware revisions behave identically.

One caveat: this doesn't solve key retrieval for already-provisioned scales. The Fitbit app normally retrieves the device-specific key through its API. The approach tested here requires resetting the scale and provisioning it with a new key.

I haven't implemented an openScale driver, but figured I'd post the protocol details here since the previous investigation got stuck on decoding the measurements.

It looks like local-only Aria Air support is feasible, including initial provisioning, without needing the Fitbit service.

ELY3M, 2026-10-26:
I am fine with resetting my scale as long I can reconnect it with openscale.  I still have the fitbit scale.  I have got different scales that is working with openScale.