oliexdev/openScale #1109: Fraudulent app

zakjakub, 2025-01-29:
It's fraudulent app. It doesn't work and they will not refund you money.

ikanakova, 2025-01-29:
I have the app from F-Droid and it works with the Beurer BF700 scale perfectly.

BenBarat, 2025-02-02:
Wonder where you bought it? How much did you pay?
What version number do you have?
What scales do you have? Make and model please.

Blue-Sauce, 2025-02-07:
> It's fraudulent app. It doesn't work and they will not refund you money.

1. The app is not guaranteed to support every scale, or the particular scale you have. Even if it has a model number that has been reported as working that is no guarantee, as manufacturers can change software or hardware without properly changing the advertised model number

2. You never needed to pay for the app. It's available right here on github as well as F-droid for free. The google play price basically is just to support the devs, if you wish to.

zakjakub, 2025-02-09:
I paid for sync app on Google Play. It doesn't work, it's blocked by Google as dangerous app. Google refused to refund money, and there is no any response to e-mails to developer.

ikanakova, 2025-02-09:
> I paid for sync app on Google Play. It doesn't work, it's blocked by Google as dangerous app. Google refused to refund money, and there is no any response to e-mails to developer.

This is already being solved here #1106, but the fact is that the developer hasn't responded to it yet.

BenBarat, 2025-02-09:
I feel your frustration. Methinks Google moved the goalposts and all the developers are frantically scrambling to catch up, or have shrugged their shoulders and moved on in life.

Look around - this is not the only program accessing Google Fit that has struck problems.

For some changes a quick update to the latest compiler version, refresh recompile  and an updated release is all that is required. For other changes, a complete fundamental re-write is needed. If you are a poor hobbyist developer (most of us are) and it costs more money to pay for an update to our development tools than what you are earning (that is why everybody is moving to a subscription model where you pay forever to use your tools) , you get a little disenchanted. If you wrote a program that solved your immediate needs and then decided to share it so others could benefit - bravo. If your now functioning device and software needs have been met, why add more energy to update it for others?

I'm not even sure if you are complaining about the (free) app here (openScale) or the (paid) sync app (last updated on Google Play in August 2023) - never got a response to my question of a few days ago of what version you are running and on what platform? Your cry of fraud was a little short on specifics.

I do not think this is the appropriate forum to be complaining about Google changing their APIs and access methods, billing procedures, software classification on Google play, and blaming a third party developer here that is forced to comply for their product to survive on that platform. I note nearly all complaints on Google forums in relation to this FitBit issue have been forcibly closed without any response.

Coming here and DEMANDING a response FORTHWITH is maybe also a little over the top. Crying fraud is a little strange - acknowledge that software becomes 'stale' and needs updating.
Github is full of abandonware. Often a pet project gets less attention as life intervenes to rearrange priorities. A product that hasn't sold well may not be motivating to add insult to injury to pour more effort to support something when users didn't support it. This is the usual bane of open source. Look at a subsequent product the same developer has released - openWorkouts with only a 100+ downloads category - hardly a best seller.

Maybe the developer does this in his spare time, and is busily re-coding his app, again in his spare time behind the scenes, (the free version which by-the-way works with two brands of my scales without issues for many years) to integrate with the Google ecosystem - a moving target.

I don't see a lot of users recently contributing their bluetooth device  profile settings for assisting to support emerging weight scale models. Either the generic bluetooth device profile is widely supported with new devices, or the developer has lost interest.  Maybe a different programming model of 'slot in' parameters where your device profile is matched to the software is required. Reverse engineering protocols and undocumented APIs is a challenging art, but boring once you have done it a few times. Chinese hardware manufacturers often use whatever firmware works with their product, as long as the cash flows in. Strict bluetooth protocol specifications are often ignored. This can be frustrating when you are trying to address a moving target.

It's open source - fork it and roll your own. Reverse engineer your own device and make your millions, dancing around Google Play, FitBit, and Android ever changing requirements.

Maybe go back and whine in the Google Play reviews where this discussion belongs, and leave this for the technical stuff? You did read the reviews there before parting with your money, didn't you? Add some information rre that the developer can go on the be able to move their product forward and see where it ends up.
Rant mode off...

oliexdev, 2025-03-09:
This app is a personal hobby project that I have worked on in my free time. The app does not engage in any fraudulent activities.

Regarding refunds, Google handles all transactions and refunds through the Play Store. Developers do not have control over this process, so any refund requests should be directed to Google. Btw that is a reason why I pay a small amount to Google and if you look closely it offered by Google Commerce Ltd and not by me.

Additionally, Google has announced that Google Fit will no longer be supported, which impacts the functionality of apps relying on it. Given this, I have already rewritten the entire openScale sync app in Kotlin to work with Health Connect but have not yet decided whether / where / when / how to publish it or not.

But I will unpublish my apps on Google Play.

closed 2025-03-09 by oliexdev

zakjakub, 2025-03-09:
Google refuses to refund money and refers to app support. Simply refund the money, the app is not working as advertised. 

[image]

TheRedWolf, 2025-03-27:
> Additionally, Google has announced that Google Fit will no longer be supported, which impacts the functionality of apps relying on it. Given this, I have already rewritten the entire openScale sync [...]

Personally I'd be fine with Health Connect replacing the Google Fit functionality, especially as Google Fit could then pull the relevant stats from Health Connect.

The app is legitimate, not fraudulent... Google just moved the goalposts.