oliexdev/openScale #1119: Image for Xiaomi Mi Scale v2 Incorrect

FrenchToucan, 2025-03-26:
The image for the Xiaomi Mi Scale v2 in the wiki actually shows the Xiaomi Body Composition Scale s400 which is not supported.

https://github.com/oliexdev/openScale/raw/master/docs/scales/mi_scalev2.jpg

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://github.com/oliexdev/openScale/wiki/Supported-scales-in-openScale
2. Scroll down to Xiaomi Mi Scale v2
3. Actual image should look like https://www.mi.com/global/product/mi-smart-scale-2/

andreas-hartmann, 2025-03-27:
Yep, I literally just fell for this, buying the S400 only learning that it is a different and unsupported model.

oliexdev, 2025-03-29:
Well back in the days I bought it under the name Mi Body Composition Scale 2 --> https://buy.mi.com/de/item/3200800037

But I am not the marketing manager of Xiaomi what they called now v2 but I change the name to Mi Body Composition Scale 2  in the wiki

FrenchToucan, 2025-03-31:
> Well back in the days I bought it under the name Mi Body Composition Scale 2 --> https://buy.mi.com/de/item/3200800037
> 
> But I am not the marketing manager of Xiaomi what they called now v2 but I change the name to Mi Body Composition Scale 2 in the wiki

@oliexdev Sounds like you're saying the S400 is really supported but under a different name. The app says it does not support the S400, though. Can you make it so it is recognized as the same device as the Mi Body Composition Scale 2?

ikanakova, 2025-03-31:
[image]

FrenchToucan, 2025-03-31:
After looking at the above chart, I want to add the clarification that I bought the S400 under the product name Mi Body Composition Scale 2. In other words, it worked for @oliexdev but it isn't working for me and @andreas-hartmann because the app is saying the S400 is not a supported device.