oliexdev/openScale #1120: Improve documentation with model recommendations

andreas-hartmann, 2025-03-27:
**Is your feature request related to a problem? Please describe.**
As someone who is specifically interested in a scale supported by this App, it would be very helpful to know which ones are better than the others. Currently there is only a very basic and coarse "features supported with this scale"-table.

**Describe the solution you'd like**
The developers of this App are uniquely positioned to rate at least some of the supported scales. Many hardware adjacent projects have recommendations for specific models or detailed tables so a user can make an informed choice, e.g. OpenWRT.

I suggest identifying known "great scales" that are supported and giving them a badge in the table, or adding a couple more columns with the exact features they have (body fat, heart rate...) and an anecdotal column ("excellent display", "best in class measuring speed"). These don't have to be full on review quality, but any added information will make it easier to make an informed decision.

**Describe alternatives you've considered**
Manually researching every supported available model, or just blindly buying one are terrible options.

**Additional context**
-

mo-g, 2025-04-26:
It would be really helpful for me as well. Since #1015 I still haven't been able to get the Medisana BS 444 to work consistently even for weight recordings and body analysis measurements are just a non-starter. Yet it's still listed as fully supported in the Wiki despite just not working haha!

I can see people talking about Sanitas SBF70 but I'm nervous to buy one following my experience with the Medisana. So; I want to use the app, (which I did donate to, for what it's worth) but I just have no way to reliably obtain any actual hardware to actually get the data.

There are 20+ body analysis scales on the Wiki but I have no way to know which ones work.

lundyfpv, 2025-06-02:
I came looking for which scales are the best supported.

oliexdev, 2025-09-13:
This project is a hobby project and I’m not affiliated with any scale manufacturer. I don’t own or test every supported scale, and unfortunately some devices have poor firmware or other limitations that openScale cannot “fix.” Because of this, I can’t guarantee 100 % functionality for any model, nor is this an official replacement or support channel for any manufacturer.

Most of the scale integrations and Wiki entries are created and maintained by the community, based on their own devices and feedback. Also, please note that some scales share the same model name but come with completely different firmware versions — especially lower-cost scales from China — and that can lead to unpredictable differences in behavior without  even noticing.

To avoid creating unrealistic expectations, I’m even considering reducing or removing the “supported scales” list entirely.

If something doesn’t work as expected, please feel free to open an issue or — even better — send a PR to improve support or documentation. That’s the main advantage of an open project like this compared to a manufacturer’s closed system.

closed 2025-09-13 by oliexdev

mo-g, 2025-09-16:
> To avoid creating unrealistic expectations, I’m even considering reducing or removing the “supported scales” list entirely.

I think that's the sensible outcome here. If there's no hardware available that can be trusted works for this - then documenting any kind of "support" for devices is a bad idea - people are always going to look at that list to see what to buy; and be frustrated that it doesn't work when they purchase something. Appreciate that this is a hobby project, and that the Google Play purchase links seem to have been pulled now removing any previous perception of this as a commercial product.

Better to focus on the DIY option which you've created! :)