oliexdev/openScale #1123: App crashes while connecting to scale via bluetooth

linkwithkk, 2025-04-20:
**Describe the bug**
App version 2.5.3
It detects the scale as QN-Scale. While connecting to the scale it crashes and asks to raise a bug here.

**To Reproduce**
Steps to reproduce the behavior:
1. Open app
2. Click on  Bluetooth icon
4. App shows connecting to QN-Scale
5. App crashes

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): Yes. No error in dev build but app does not records any reading and keeps showing message as "please step barefoot on scale". No data is ever recorded.

**Expected behavior**
App should take a measurement.

**Additional context**
NA

**Debug log**

Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-04-20 19:49:41 
Device: Samsung SM-F741B 
OS version: Android 14 (SDK 34) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:230)
	at android.os.Looper.loop(Looper.java:319)
	at android.app.ActivityThread.main(ActivityThread.java:9063)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:588)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)

linkwithkk, 2025-04-20:
Just to add, when I tried with app fitindex it worked fine. Let me know if I need to send any other information

wanderer, 2025-04-21:
I'm also having this problem
```
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-04-18 08:07:55 
Device: Google Pixel 9 Pro 
OS version: Android 15 (SDK 35) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:991)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8934)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)
```

g4rb4g3, 2025-04-29:
Did you try the latest dev build?
https://github.com/oliexdev/openScale/releases/tag/dev-build