oliexdev/openScale #1124: 🎉 openScale sync Beta Testing Invitation!

oliexdev, 2025-05-10:
Hello openScale community!

I'm excited to announce that I've recently completed a significant rewrite of the openScale sync application in Kotlin! This rewrite brings performance improvements and a more modern codebase.

Before releasing the updated app (version 0.4) to production on Google Play, I'm looking for **at least 12 beta testers** to help test the application and provide valuable feedback. Your participation is crucial to ensure a smooth and stable production release.

Beyond just improving the app through your valuable feedback, conducting a closed beta test with a minimum number of testers is a requirement by Google Play for publishing production releases. Your participation directly helps meet these requirements and brings us closer to making the improved openScale sync available to everyone!

**What's New in Version 0.4?**

*   **Complete rewrite in Kotlin:** Modernized codebase for improved performance and maintainability.
*   **Health Connect Integration:** You can now seamlessly sync your openScale measurements with Android's Health Connect! 

**What you'll be testing:**

You will be testing the core functionality of openScale sync, including:

*   Setting up and synchronizing with external services (Health Connect, Wger, MQTT).
*   Real-time synchronization of measurements.
*   Manual full synchronization.
*   Configuring service activation and user selection.
*   The overall stability and user experience of the app.

**How to become a beta tester:**

If you have a **Google account (Gmail address)** and are willing to actively test the openScale sync beta version and provide feedback, please **send me an e-mail at olie.xdeveloper@googlemail.com or comment on this issue** expressing your interest.

**Important:** Please include the **Gmail address** you use for your Google Play account in your comment. I will use this to invite you to the beta program on Google Play.

Once you are invited and accept, you will be able to download and install the beta version of openScale sync through the Google Play Store.

**Providing Feedback:**

During the beta testing period, please report any bugs, crashes, or unexpected behavior you encounter by opening a new issue on this GitHub repository. Please include as much detail as possible, including:

*   Device model and Android version.
*   Steps to reproduce the issue.
*   Any relevant error messages or screenshots.

You are also welcome to share any suggestions for improvements or general feedback.

Thank you in advance for your willingness to help make openScale sync even better! Your contributions are highly valued.

Best regards,
olie.xdev

TheRedWolf, 2025-05-10:
I'm in! Sent an email...

Foguet, 2025-05-11:
I have already sent an email to test the beta...

kub3let, 2025-05-21:
dammit just yesterday i wrote a sync myself using the backup feature, now i see this issue

count me in, looking forward to live sync using mqtt

can you post beta builds here or publish it on f-droid ? not using google

cmmata, 2025-05-22:
Hi! I'm an OpenScale pro user. I have just bought a new phone, and when I tried to copy it to the new phone I realized that the app does not exist on Google Play anymore. I can only install it from F-droid, losing the pro buy. I can help to test the sync beta, but will it become a separate paid app? Or the sync feature will be included on the regular (free) app?

oliexdev, 2025-05-22:
openScale sync will be free but currently I need more beta tester, so everybody is welcome to test it. Please send me a e-mail with your Google e-mail address so I can add you as a beta tester.

kpadmanabhan, 2025-05-29:
@oliexdev  I am in. Have sent you an email in this regard.

KonradIT, 2025-06-05:
Hi, been testing the following app versions:

- openScale dev (md5: 9b6ac26429e29eb2c27882b86c1ddb38)
- openScale sync from play store: 0.4-oss (10)

When going on the scale and sending a measurement to openScale, it doesn't automatically get sent over to the sync app.

See:

[image]

For June 5, I logged 2 measurements 

Yet the last time it was synced to Health Connect its on 4th June:

[image]

Seems I still have to manually click the full sync button for that to happen. 

I made sure to allow read/write OpenScale data permission as well as health connect sync permission in settings.

hitesh-singh, 2025-06-05:
> [@hitesh-singh](https://github.com/hitesh-singh) I don't mind sharing but it's the most ugliest code ever... https://github.com/kub3let/Esp32OpenScaleSync
> 
> Also it only works for the SBF75, I only implemented the protocol parts to get measurement data from https://github.com/oliexdev/openScale/wiki/Beurer-Sanitas

Thank much @kub3let, It's gives me a very good starting to work on for my needs 🍻

oliexdev, 2025-06-06:
> Hi, been testing the following app versions:
>
> * openScale dev (md5: 9b6ac26429e29eb2c27882b86c1ddb38)
> * openScale sync from play store: 0.4-oss (10)
>
> When going on the scale and sending a [...]

It should be synced, please check the HealhtConnect App if the data has been synced. The last sync timestamp is only used when you full sync the data. It is misseading I will change that.

Nefari0uss, 2025-06-20:
Are you still looking for testers?

oliexdev, 2025-06-21:
@Nefari0uss sure please send me your googlemail e-mail address so I can add you as a beta tester

oliexdev, 2025-06-25:
Thanks everybody for testing and your valuable feedback. I have fixed some issues and released the openScale sync 0.4.1 app at [Google Play](https://play.google.com/store/apps/details?id=com.health.openscale.sync.oss) for free.

I also open-sourced openScale sync at https://github.com/oliexdev/openScale-sync so everbody is welcome to contribute to the project. 

Last but not least, I have requested it for publishing on fdroid, more to come.