oliexdev/openScale #1126: Crashes when trying to connect to Renpho scale

rdnielsen, 2025-05-21:
**Describe the bug**
An error message is displayed with diagnostics when stepping on the scale.

**To Reproduce**
Steps to reproduce the behavior:
1. Open app.
2. Enable Bluetooth
3. Step on scale
4. See error

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): 2.5.3 (65)

**Expected behavior**
App acquires data from scalee

**Additional context**
Diagnostics:
Build version: 2.5.3 
Build date: 1981-01-01 01:01:02 
Current date: 2025-05-21 07:11:39 
Device: Samsung SM-X210 
OS version: Android 14 (SDK 34) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(Unknown Source:5)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothQNScale.onNextStep(BluetoothQNScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:958)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loopOnce(Looper.java:230)
	at android.os.Looper.loop(Looper.java:319)
	at android.app.ActivityThread.main(ActivityThread.java:8934)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:578)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1103)

**Debug log**
Attach a debug log (see Settings -> About -> Save debug log to file) that is captured while reproducing the issue.

maquis196, 2025-05-24:
Not sure if +1 or thumbsup preferred, but I hit this earlier too.

reversesh3ll, 2025-06-10:
Same issue for me as well on my ES-CE20M.

gramss, 2025-07-11:
Same issue also for me with latest 2.5.4

```
Build version: 2.5.4 
Build date: 1981-01-01 01:01:02 
Current date: 2025-07-11 18:30:29 
Device: Samsung SM-A405FN 
OS version: Android 11 (SDK 30) 
 
Stack trace:  
java.lang.NullPointerException: no valid characteristic provided
	at j$.util.Objects.requireNonNull(SourceFile:244)
	at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
	at com.health.openscale.core.bluetooth.BluetoothRenphoScale.onNextStep(BluetoothRenphoScale.java:111)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
	at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
	at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
	at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
	at android.os.Handler.handleCallback(Handler.java:938)
	at android.os.Handler.dispatchMessage(Handler.java:99)
	at android.os.Looper.loop(Looper.java:246)
	at android.app.ActivityThread.main(ActivityThread.java:8653)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:602)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1130)
```

keeperofdakeys, 2025-07-20:
I just hit this same issue, with my crashdump showing this line:

"	at com.health.openscale.core.bluetooth.BluetoothRenphoScale.onNextStep(BluetoothRenphoScale.java:111)"

My Renpho scale is a ES-CS20M. I would expect it to use the BluetoothESCS20M class, but instead it's using the BluetoothRenphoScale class. This suggests a bug in the scale detection code.

I suspect this is happening because my scale is reporting a bluetooth name of "Renpho-Scale".

Debug logs show:

Found supported device Renpho-Scale [FF:04:00:13:XX:XX] (Driver: RENPHO ES-WBE28)

Suggestions:
1. Can we identify scales via bluetooth calls to scale?
2. Can we identify scales via mac address?
3. Add feature to override scale model?

PS: Sounds like there are two bugs here, since some logs are referencing BluetoothQNScale.

gramss, 2025-07-20:
nice find @keeperofdakeys .. Just as a reference, here is the commit adding the ES-CE20M to the code stack: https://github.com/oliexdev/openScale/commit/c2d29db66d897d73a02f080846f5c4720709cd4a

And by that I did not only found the responsible code for identifying the scale (like you suggested) but also found a comment that would hint exactly to that kind of potential issue in the future with `RenphoScales` 🔥 😀 https://github.com/oliexdev/openScale/blob/72d748292abcadb7e88bb1f3abdbad9fef79d456/android_app/app/src/main/java/com/health/openscale/core/bluetooth/BluetoothFactory.java#L159

Maybe we could just swap the detection of the `ES-CS20M` above the generic `RenphoScales` detection and call it a day?
I am just not too familiar with the whole android ecosystem..

gramss, 2025-07-20:
Yes, you are right.

Just checked. Via a BLE scanner, I also noticed that my scaled is using the generic provisioning name `Renpho-Scale`.
So swapping the name matching order would fix nothing.

That brings me to the idea that one could encapsulate the first matching with the "wrong" protocol into a try-catch clause.. But that probably is messing the connection/pairing routing if the first part is parsed incorrectly.

The matching part of the BluetoothFactory currently only forwards the name of the device. So forwarding the `mac-address` to the BluetoothFactory and working on that could help?

If it helps, that is my `Manufacture Data` of my scale:
ff-ff-09-5b-01-00-00-b5-09-XX-00-XX-XX-XX-XX-XX-XX-XX

Would you mind sharing yours?
And yes, you are right that they might also changed the firmware and not only the name of the BLE broadcasting device..

keeperofdakeys, 2025-07-23:
I've taken some dumps, and this scale seems more like BluetoothQNScale. With the official app I'm only seeing chateristic fff1 and service fff0, so doesn't look like BluetoothESCS20M or BluetoothRenphoScale.

I'll keep looking at this and see how I go.

gramss, 2025-07-26:
Thanks @keeperofdakeys !
Do you need help digesting the BLE dumps?

keeperofdakeys, 2025-07-27:
I've made some progress. The BluetoothQNScale class was missing some code needed for this revision of the protocol. Some was commented out (copied from a different implementation), some I had to add. For testing I've modified the app to use BluetoothQNScale for this scale, and I can see the app receiving the measurements in the debug log. However no data appears in the app.

So some good progress, but some work still to go.

gramss, 2025-07-27:
Would you mind creating a PR/branch? So I can try helping to find the missing pieces and test it 🙂

keeperofdakeys, 2025-07-29:
Alright I've got this working successfully now. Remaining work is to fix the scale selection, and do some code cleanup. Scale selection will require some changes to allow BluetoothFactory to match on BLE Service UUIDs. Once this is done I'll make a proper PR.

The code is in my repo here. https://github.com/keeperofdakeys/openScale

keeperofdakeys, 2025-07-30:
There is now a PR for this.

https://github.com/oliexdev/openScale/pull/1152