oliexdev/openScale #1138: bas contraste with dark mode

hferee, 2025-06-27:
**Describe the bug**
When using a dark mode, some of the texte in the setup instructions is hard to read as it is black on black background

**To Reproduce**
Steps to reproduce the behavior:
1. Enable dark mode on android
2. Install the app
3. Go through the setup
4. Notice the dark text

Reproduced with 2.5.3.

**Expected behavior**
All text should be clearly readable with all modes.

Kdmeenaa, 2025-07-05:
@carlesm @bittin @d3m3vilurr @hferee 
Hi maintainers 👋
 review my PR And push the main branch  Thanks! 🙏

bittin, 2025-07-05:
> [@carlesm](https://github.com/carlesm) [@bittin](https://github.com/bittin) [@d3m3vilurr](https://github.com/d3m3vilurr) [@hferee](https://github.com/hferee) Hi maintainers 👋 review my PR And push the main branch Thanks! 🙏

not a maintainer, just helped out with some Swedish translation in December 2023 via: hosted.weblate.org/languages/sv/

closed 2025-08-03 by oliexdev