oliexdev/openScale #1139: 🎉 openScale Beta Testing Invitation!

oliexdev, 2025-06-27:
Hello openScale community!

First off, a huge thank you for your invaluable help in beta testing openScale sync! Thanks to your feedback and dedication, I successfully published it on [Google Play](https://play.google.com/store/apps/details?id=com.health.openscale.sync.oss) and make it open-source at [Git Hub](https://github.com/oliexdev/openScale-sync) and hopefully will soon published on [f-droid](https://gitlab.com/fdroid/rfp/-/issues/3171).

I'm reaching out again because I am hoping I could lean on your expertise one more time. As you know, to publish the main openScale app to Google Play it requires to have at least 12 active beta testers before it can go live. Your participation directly helps meet these requirements and brings us closer to making openScale available to everyone!

**How to become a beta tester:**

If you have a **Google account (Gmail address)** and are willing to actively test the openScale beta version and provide feedback, please **send me an e-mail at olie.xdeveloper@googlemail.com or comment on this issue** expressing your interest.

**Important:** Please include the **Gmail address** you use for your Google Play account in your comment. I will use this to invite you to the beta program on Google Play.

Once you are invited and accept, you will be able to download and install the beta version of openScale through the Google Play Store.

**Providing Feedback:**

During the beta testing period, please report any bugs, crashes, or unexpected behavior you encounter by opening a new issue on this GitHub repository. Please include as much detail as possible, including:

*   Device model and Android version.
*   Steps to reproduce the issue.
*   Any relevant error messages or screenshots.

You are also welcome to share any suggestions for improvements or general feedback.

Thank you in advance for your willingness to help make openScale even better! Your contributions are highly valued.

Best regards,
olie.xdev

CaffeineOnIce, 2025-07-14:
I'd like to beta test the app, if possible. I have a smart scale, that it connects to, but has some bugs.

bsimmo, 2025-08-30:
How do you want feedback on V3 to be provided (to keep it from v2 feedback/bugs) in the Issues?

I'll add my details later, but it's working.

Feedback 1 - Icons/Avatars/selfie for different users so visually you can see who you are reading about.
Feedback 2 - Stone/lb not Stone/decimal fraction of stone (I'm from UK so switch between stone-lb and kg, but never use decimal stones)
Query 1 -  When I altered weight to Stone, about said something about changing numbers, why? what is it doing?  It should just read everything in the default unit (kg) and then use a conversion at run time for the gui preference.  It shouldn't alter the data. (this is what I had my updated version of WiiScale do). Not least because converting it back adds error.  Did I mention we switch between kg on stone regularly when using it in different places.

It's working really nice, going to have a wider play with it. thanks.

woaloo, 2025-08-30:
Hello,
I installed beta 3 to test it with the RENPHO ES-CS20M scale. The initial setup went well, and the scale is connected to the app. However, the first time I weighed in, the app crashed and closed immediately.

bsimmo, 2025-08-30:
> Create a new issue, see above "Providing Feedback" section
> good idea will keep that in mind
> I am not from UK could you describe it more in detail.
> The behavior is intentional. I keep kilograms as the canonical store, and
> when you change the unit preference I refresh cached display conversions
> (e.g., to stone) so large datasets don’t have to be recomputed across every
> view. The raw kg data isn’t altered; round-trip conversions use precise
> factors, so any drift beyond UI rounding would be a bug—please share an
> example if you see one. That said, I’m a bit surprised the units are
> switched so often—could you explain why you regularly toggle between kg and
> Please provide the log file but could be possible that this issue is a
> scale problem, search for it in the issue.

I will open up issues to open up discussions, the last part isn't me
(RENPHO). Mine works fine.

gassssss, 2025-09-03:
Hi, I'm already testing the beta through downloading through github directly, so far is OK. This is the second time I try and it imported the old db without issues, recognized the scale (Mengii) and grabbed the data correctly.

The reason I went back to the stable version (and probably will go back again) is that idk why every time I try to update the beta (by github download/install) I get the message that it conflicts with an existing package, forcing to uninstall-reinstall-reconfigure-import which gets annoying after a few times. Not sure if it is a skill issue or what, but I have a degoogled phone can't/won't join the GPlay test.

Nonetheless is looking great so far, thanks a lot for your work.

oliexdev, 2025-09-05:
Please open a new issue but please test it with the latest beta build on GitHub

oliexdev, 2025-10-18:
> Hi, I tested the latest beta with my scale ([#1081](https://github.com/oliexdev/openScale/pull/1081)), but I had to do additional changes to restore the support for nameless devices. The changes are here : [Florin9doi@d7aeb76](https://github.com/Florin9doi/openScale/commit/d7aeb76a13de31b2eadb1ef3ddc1c87fd3f799b0) . Beside this, I found few obstacles in comparison with the stable release:

Thanks I have just commited a change that the manufactured data is also stored when you save the device, please try out again then we don't need a special handling here.

> 
> * No support for Android <12. This isn't something I need, I just tried with no luck to install it on an old phone to avoid it messing with the stable app on my primary phone.

The latest supported SDK is 31, which will make thinks easier for e.g. permission handling.

> * No delete confirmation.

Yeah it is still on my todo list.

> * The new version features a single fractional digit, resulting in slightly different values than those displayed on the scale.

I think the precision with one digit is enough for a body scale, if the scale will show more than one digit than I would questions the scale calculation.

oliexdev, 2025-10-19:
> No delete confirmation.

@Florin9doi should now be implemented in the latest beta build

oliexdev, 2025-10-19:
The beta version are currently heavily on development that it makes no sense to update everytime on GooglePlay. In the log file you see the git commit hash which tell you exactly the build.

Florin9doi, 2025-10-26:
>Thanks I have just commited a change that the manufactured data is also stored when you save the device, please try out again then we don't need a special handling here.

It works. I removed that part and added support for Jin unit: https://github.com/Florin9doi/openScale/commit/4712f89f2e3d850791ae23247557de802d80b9b6

>I think the precision with one digit is enough for a body scale, if the scale will show more than one digit than I would questions the scale calculation.

In beta version the trailing zeros of the fractional part are hidden anyway. It would have no impact for the scales with 100g precision.

nistvan86, 2025-10-30:
How can I opt out from the Beta? I bought a new phone (Samsung S25+ with Android 16/OneUI 8), trying to move over but I'm unable to sync with my Xiaomi Mi Scale, no matter what I do.
The previous version (v2.5.3-pro) works just fine on my S21 FE still.

I get this "Failed to enable notifications" error and it just stops there.
I've removed the batteries from the scale and try to reset it, but didn't help.

[image]

oliexdev, 2025-11-02:
@nistvan86 use the latest beta version on GitHub and attached the log file

@chainria the eye symbol was only for showing the measurement in detail which you can already do in the beta version.

hackel, 2025-11-08:
This looks great, and I would love to test the 3.0 beta, but from comments on this issue it sounds like it uses the same application ID and can't be installed side-by-side with 2.x.  Is there any chance of publishing a separate build with a unique application ID?  I don't want to risk losing or corrupting my existing 2.x data in any way, including the data backed up to Google's servers.  Thanks!

Florin9doi, 2025-11-08:
> This looks great, and I would love to test the 3.0 beta, but from comments on this issue it sounds like it uses the same application ID and can't be installed side-by-side with 2.x. Is there any [...]

They have different IDs. You can install both side by side.

bsimmo, 2025-11-08:
This.
You install and work side by side.
Every time you update V3 beta you need to uninstall the V3, then install
the updated V3.
(That might be what you are reading about).

oliexdev, 2025-11-16:
Thanks everbody for testing openScale 3.0 🥇 I have finally merged the openScale 3.0 beta branch into the master branch. I will step by step migrate 2.x into 3.0

closed 2025-11-16 by oliexdev