oliexdev/openScale #1160: Beuerer BF 950 on oss-3.0 (67)

chriszuercher, 2025-08-21:
**Describe the bug**
I tried my BF 950 using the latest beta from Play store. I cannot receive a measurement as the users of the scale cannot be read / selected

**To Reproduce**
Steps to reproduce the behavior:
1. Click on Bluetooth icon
2. Wait for scale to be connected
4. See error

[image]

Reproduced with 3.0-oss (67)

**Expected behavior**
A user can be selected and my measurement received.

**Debug log**

[openScale_current_log.txt](https://github.com/user-attachments/files/21914690/openScale_current_log.txt)

oliexdev, 2025-08-21:
Should be already be fixed in the latest beta build on GitHub, see https://github.com/oliexdev/openScale/issues/1156

chriszuercher, 2025-08-26:
I downloaded the latest build from Github. Now the user selection works, and I'm asked for a user. On the first try I had one user, then somehow they duplicated, and now I have 4 of them. Also, I'm asked for a "Confirmation Code" which I have no idea how to get it. So for now I cannot get any data from my scale. Do you see why?

[image]

[image]

[openScale_current_log.txt](https://github.com/user-attachments/files/21988682/openScale_current_log.txt)

bsimmo, 2025-08-30:
Could you document the PIN as I have this problem, also use another phone and try and connect to that user, it keeps asking for PIN #1149 
Resetting just wipes out all users and you start again, not ideal as a few of use use the scale.

https://github.com/oliexdev/openScale/pull/753#issuecomment-3198013555