oliexdev/openScale #1166: [Bug] RENPHO ES-CS20M crash on measurement (OpenScale 2.5.4 / Android 15)

woaloo, 2025-08-30:
After connecting the RENPHO ES-CS20M to OpenScale 2.5.4, the app crashes as soon as I step on the scale.

Environment:
OpenScale version: 2.5.4
Device: Fairphone FP5
Android: 15 (SDK 35)
Balance: RENPHO ES-CS20M

Steps to reproduce:
1. Pair scale via Bluetooth => OK
2. Open OpenScale => OK
3. Step on the scale 
4. App crashes

Expected: weight should be recorded
Actual: app crashes with NullPointerException

Stack trace :
Build version: 2.5.4
Build date: 1981-01-01 01:01:02
Current date: 2025-08-29 19:35:41
Device: Fairphone FP5
OS version: Android 15 (SDK 35)

Stack trace: 
java.lang.NullPointerException: no valid characteristic provided
at j$.util.Objects.requireNonNull(SourceFile:244)
at com.welie.blessed.BluetoothPeripheral.writeCharacteristic(BluetoothPeripheral.java:1224)
at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:303)
at com.health.openscale.core.bluetooth.BluetoothCommunication.writeBytes(BluetoothCommunication.java:291)
at com.health.openscale.core.bluetooth.BluetoothRenphoScale.onNextStep(BluetoothRenphoScale.java:111)
at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:610)
at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
at com.health.openscale.core.bluetooth.BluetoothCommunication.nextMachineStep(BluetoothCommunication.java:612)
at com.health.openscale.core.bluetooth.BluetoothCommunication.resumeMachineState(BluetoothCommunication.java:210)
at com.health.openscale.core.bluetooth.BluetoothCommunication$1.onServicesDiscovered(BluetoothCommunication.java:464)
at com.welie.blessed.BluetoothPeripheral$1$1.run(BluetoothPeripheral.java:213)
at android.os.Handler.handleCallback(Handler.java:991)
at android.os.Handler.dispatchMessage(Handler.java:102)
at android.os.Looper.loopOnce(Looper.java:232)
at android.os.Looper.loop(Looper.java:317)
at android.app.ActivityThread.main(ActivityThread.java:8934)
at java.lang.reflect.Method.invoke(Native Method)
at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)

woaloo, 2025-08-31:
Hi,
I have the same issue with the Beta 3
When I try to add a measurement, the app crash and quit.
Here is the log.

| Field | Value |
|---|---|
| Time | 2025-08-31 09:11:51.138 |
| Session ID | 198fef7d362 |
| App | openScale |
| Version | 3.0-beta (68) |
| Package | com.health.openscale.beta |
| Build Type | beta |
| Device | Fairphone FP5 |
| Android | 15 (API 35) |
| Build ID | BP1A.250505.005 release-keys |

```diff
? 2025-08-31 09:11:51.139 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.beta/files/logs/openScale_current_log.txt
+ 2025-08-31 09:11:51.139 I/openScaleLog: File logging enabled during runtime – started fresh session log.
. 2025-08-31 09:12:07.524 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 09:12:07.525 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 09:12:07.559 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 09:12:07.560 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 09:12:07.566 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 09:12:07.566 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 09:12:07.589 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 09:12:07.589 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 09:12:07.603 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 09:12:07.603 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 09:12:07.626 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 09:12:07.626 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
+ 2025-08-31 09:12:10.539 I/BluetoothScannerMgr: Closing BluetoothScannerManager.
+ 2025-08-31 09:12:10.542 I/BluetoothConnManager: Closing BluetoothConnectionManager.
? 2025-08-31 09:12:10.543 D/BluetoothConnManager: Closing manager: Releasing active communicator: null
? 2025-08-31 09:12:10.542 D/BluetoothScannerMgr: Blessed BluetoothCentralManager closed successfully.
? 2025-08-31 09:12:10.544 D/BluetoothConnManager: Closing manager: Active communicator released and set to null.
+ 2025-08-31 09:12:10.545 I/BluetoothConnManager: BluetoothConnectionManager closed.
? 2025-08-31 10:42:15.654 D/MainActivity: Observed language code: null, Current activity locale: en, Target: en
? 2025-08-31 10:42:15.654 D/MainActivity: Initializing and setting content.
. 2025-08-31 10:42:15.727 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 10:42:15.728 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 10:42:15.810 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 10:42:15.811 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 10:42:15.856 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 10:42:15.857 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
? 2025-08-31 10:42:15.907 D/SharedViewModel: Backfill skip: no measurements for userId=1.
+ 2025-08-31 10:42:15.907 I/SharedViewModel: Derived backfill not needed for any user.
. 2025-08-31 10:42:15.918 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 10:42:15.919 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 10:42:15.926 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 10:42:15.927 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 10:42:15.950 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 10:42:15.951 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 10:42:15.955 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-08-31 10:42:15.956 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-08-31 10:42:15.982 V/UserSettingsRepository: Setting 'overview_screen_show_type_filter_row' not found, returning default value: true
. 2025-08-31 10:42:15.983 V/UserSettingsRepository: Setting 'overview_screen_time_range' not found, returning default value: ALL_DAYS
. 2025-08-31 10:42:27.613 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
[... cut ...]
? 2025-08-31 10:42:31.354 D/BluetoothCommunication: Resume machine state
? 2025-08-31 10:42:31.356 D/BluetoothCommunication: Step Nr 0
+ 2025-08-31 10:42:31.358 I/BluetoothRenphoScale: onNextStep(0)
? 2025-08-31 10:42:31.361 D/BluetoothCommunication: Invoke set notification on 0x2a2b "current time"
? 2025-08-31 10:42:31.363 D/BluetoothCommunication: Step Nr 1
+ 2025-08-31 10:42:31.366 I/BluetoothRenphoScale: onNextStep(1)
? 2025-08-31 10:42:31.368 D/BluetoothCommunication: Invoke set indication on 0x2a9f "user control point"

woaloo, 2025-09-03:
Hi,
Thank you for youe app and your support.
I've test with the latest one. Good news, no crash !
Bad news :

[image]

I have enable logging, but can't find where it is

woaloo, 2025-09-04:
| Field | Value |
|---|---|
| Time | 2025-09-03 20:24:20.956 |
| Session ID | 19910d297dc |
| App | openScale |
| Version | 3.0-beta (68) |
| Package | com.health.openscale.beta |
| Build Type | beta |
| Device | Fairphone FP5 |
| Android | 15 (API 35) |
| Build ID | BP1A.250505.005 release-keys |

```diff
? 2025-09-03 20:24:20.957 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.beta/files/logs/openScale_current_log.txt
+ 2025-09-03 20:24:20.957 I/openScaleLog: File logging enabled during runtime – started fresh session log.
. 2025-09-03 20:24:31.977 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-09-03 20:24:31.977 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-09-03 20:24:32.002 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-09-03 20:24:32.003 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
. 2025-09-03 20:24:32.023 V/UserSettingsRepository: Observing setting: key='overview_screen_time_range', type='String'
. 2025-09-03 20:24:32.023 V/UserSettingsRepository: Observing setting: key='overview_screen_show_type_filter_row', type='Boolean'
+ 2025-09-03 20:24:52.536 I/BluetoothScannerMgr: Closing BluetoothScannerManager.
- 2025-09-03 20:24:52.539 E/BluetoothScannerMgr: Error closing Blessed BluetoothCentralManager: Receiver not registered: com.welie.blessed.BluetoothCentralManager$11@e861cc2
java.lang.IllegalArgumentException: Receiver not registered: com.welie.blessed.BluetoothCentralManager$11@e861cc2
	at android.app.LoadedApk.forgetReceiverDispatcher(LoadedApk.java:1685)
	at android.app.ContextImpl.unregisterReceiver(ContextImpl.java:1958)
	at android.content.ContextWrapper.unregisterReceiver(ContextWrapper.java:843)
	at com.welie.blessed.BluetoothCentralManager.close(BluetoothCentralManager.java:329)
	at com.health.openscale.core.service.BluetoothScannerManager.close(BleScanner.kt:237)
	at com.health.openscale.core.facade.BluetoothFacade.close(BluetoothFacade.kt:222)
	at com.health.openscale.ui.screen.settings.BluetoothViewModel.onCleared(BluetoothViewModel.kt:126)
	at androidx.lifecycle.ViewModel.clear$lifecycle_viewmodel_release(ViewModel.jvm.kt:68)
	at androidx.lifecycle.ViewModelStore.clear(ViewModelStore.kt:58)
	at androidx.activity.ComponentActivity._init_$lambda$3(ComponentActivity.kt:271)
	at androidx.activity.ComponentActivity.$r8$lambda$ibk6u1HK7J3AWKL_Wn934v2UVI8(Unknown Source:0)
	at androidx.activity.ComponentActivity$$ExternalSyntheticLambda2.onStateChanged(D8$$SyntheticClass:0)
	at androidx.lifecycle.LifecycleRegistry$ObserverWithState.dispatchEvent(LifecycleRegistry.jvm.kt:313)
	at androidx.lifecycle.LifecycleRegistry.backwardPass(LifecycleRegistry.jvm.kt:266)
	at androidx.lifecycle.LifecycleRegistry.sync(LifecycleRegistry.jvm.kt:284)
	at androidx.lifecycle.LifecycleRegistry.moveToState(LifecycleRegistry.jvm.kt:135)
	at androidx.lifecycle.LifecycleRegistry.handleLifecycleEvent(LifecycleRegistry.jvm.kt:119)
	at androidx.lifecycle.ReportFragment$Companion.dispatch$lifecycle_runtime_release(ReportFragment.android.kt:181)
	at androidx.lifecycle.ReportFragment$LifecycleCallbacks.onActivityPreDestroyed(ReportFragment.android.kt:137)
	at android.app.Activity.dispatchActivityPreDestroyed(Activity.java:1768)
	at android.app.Activity.performDestroy(Activity.java:9385)
	at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1570)
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:6108)
	at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:6153)
	at android.app.servertransaction.DestroyActivityItem.execute(DestroyActivityItem.java:52)
	at android.app.servertransaction.ActivityTransactionItem.execute(ActivityTransactionItem.java:63)
	at android.app.servertransaction.TransactionExecutor.executeLifecycleItem(TransactionExecutor.java:169)
	at android.app.servertransaction.TransactionExecutor.executeTransactionItems(TransactionExecutor.java:101)
	at android.app.servertransaction.TransactionExecutor.execute(TransactionExecutor.java:80)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2773)
	at android.os.Handler.dispatchMessage(Handler.java:109)
	at android.os.Looper.loopOnce(Looper.java:232)
	at android.os.Looper.loop(Looper.java:317)
	at android.app.ActivityThread.main(ActivityThread.java:8934)
	at java.lang.reflect.Method.invoke(Native Method)
	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:591)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:911)

+ 2025-09-03 20:24:52.541 I/BluetoothConnManager: Closing BluetoothConnectionManager.
+ 2025-09-03 20:24:52.542 I/BluetoothConnManager: disconnect() called for Debug. Active communicator: true, Status: CONNECTING
+ 2025-09-03 20:24:52.549 I/BluetoothConnManager: BluetoothConnectionManager closed.
! 2025-09-03 20:24:55.552 W/BluetoothConnManager: Disconnect timeout for Debug. Forcing status to DISCONNECTED.
? 2025-09-03 20:24:55.555 D/BluetoothConnManager: Disconnect timeout: Releasing active communicator: GattScaleAdapter
? 2025-09-03 20:24:55.560 D/BluetoothConnManager: Disconnect timeout: Active communicator released and set to null.

<img width="1224" height="2700" alt="Image" src="https://github.com/user-attachments/assets/4f66b846-9d30-4bd1-b752-41669cbc8a63" />

woaloo, 2025-09-04:
[image]

woaloo, 2025-09-04:
[image]

oliexdev, 2025-09-04:
can you set your scale into the debug mode and post the log file again?

oliexdev, 2025-09-04:
Ok you can remove the debug scale and use the correct one (select the correct scale). Try out the latest beta build and post the log file and explain step by step what are you doing and what weight value is displayed on your scale when you finished (wait if your weight is stable!)

woaloo, 2025-09-05:
OK, here hat I do.
Create a User
Grant BT permissions
Step on scale
Add BT Scale
Add mesurement
Export log file

| Field | Value |
|---|---|
| Time | 2025-09-05 12:06:42.366 |
| Session ID | 1991957b4be |
| App | openScale |
| Version | 3.0-beta (68) |
| Package | com.health.openscale.beta |
| Build Type | beta |
| Device | Fairphone FP5 |
| Android | 15 (API 35) |
| Build ID | BP1A.250505.005 release-keys |
| Build Time | 2025-09-04T18:00:48Z |
| Git SHA | 2766b20 |

```diff
? 2025-09-05 12:06:42.361 D/openScaleLog: Markdown diff block closed.
? 2025-09-05 12:06:42.367 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.beta/files/logs/openScale_current_log.txt
+ 2025-09-05 12:06:42.368 I/openScaleLog: File logging enabled during runtime – started fresh session log.
+ 2025-09-05 12:07:05.889 I/BluetoothScannerMgr: Starting device scan for 20000 ms.
+ 2025-09-05 12:07:08.326 I/BluetoothScannerMgr: Stopping device scan. Triggered by timeout: false
? 2025-09-05 12:07:08.330 D/BluetoothScannerMgr: Scan stopped. Found devices: 3
+ 2025-09-05 12:07:08.330 I/UserSettingsRepository: Saving device snapshot: addr=FF:04:00:0D:1E:70, name=Renpho-Scale, uuids=1, hint=QN Scale
+ 2025-09-05 12:07:29.582 I/BluetoothConnManager: Attempting to connect to Renpho-Scale
? 2025-09-05 12:07:29.583 D/BluetoothConnManager: Switching to new device: Releasing active communicator: null
? 2025-09-05 12:07:29.584 D/BluetoothConnManager: Switching to new device: Active communicator released and set to null.
? 2025-09-05 12:07:29.585 D/ScaleHandlerFactory: createCommunicator: Searching for communicator for 'Renpho-Scale' (FF:04:00:0D:1E:70). Handler hint: 'QN Scale'
+ 2025-09-05 12:07:29.587 I/ScaleHandlerFactory: Modern handler 'QN Scale' supports 'Renpho-Scale'.
+ 2025-09-05 12:07:29.597 I/ScaleHandlerFactory: Modern communicator 'GattScaleAdapter' created for 'Renpho-Scale'.
+ 2025-09-05 12:07:29.598 I/BluetoothConnManager: ActiveCommunicator successfully created: GattScaleAdapter. Starting observation job...
? 2025-09-05 12:07:29.601 D/BluetoothConnManager: Adapter isConnected: false for Renpho-Scale (Status: CONNECTING)
+ 2025-09-05 12:07:29.601 I/BluetoothConnManager: Adapter no longer reports connected for Renpho-Scale. Current status: CONNECTING. Expecting Disconnected Event.
+ 2025-09-05 12:07:29.704 I/GattScaleAdapter: Found FF:04:00:0D:1E:70 → stop scan + connect
? 2025-09-05 12:07:31.338 D/GattScaleAdapter: Services discovered for FF:04:00:0D:1E:70
? 2025-09-05 12:07:31.344 D/ScaleDeviceHandler: attach()
? 2025-09-05 12:07:31.346 D/ScaleDeviceHandler: handleConnected(userId=1, height=176.0, age=18)
? 2025-09-05 12:07:31.347 D/ScaleDeviceHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe1-0000-1000-8000-00805f9b34fb → false
? 2025-09-05 12:07:31.349 D/ScaleDeviceHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe2-0000-1000-8000-00805f9b34fb → false
? 2025-09-05 12:07:31.350 D/ScaleDeviceHandler: → setNotifyOn svc=0000fff0-0000-1000-8000-00805f9b34fb chr=0000fff1-0000-1000-8000-00805f9b34fb
? 2025-09-05 12:07:31.351 D/ScaleDeviceHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe3-0000-1000-8000-00805f9b34fb → false
? 2025-09-05 12:07:31.353 D/ScaleDeviceHandler: → write svc=0000fff0-0000-1000-8000-00805f9b34fb chr=0000fff2-0000-1000-8000-00805f9b34fb len=9 withResp=true payload=[13 09 15 01 10 00 00 00 42]
? 2025-09-05 12:07:31.355 D/ScaleDeviceHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe4-0000-1000-8000-00805f9b34fb → false
? 2025-09-05 12:07:31.356 D/ScaleDeviceHandler: → write svc=0000fff0-0000-1000-8000-00805f9b34fb chr=0000fff2-0000-1000-8000-00805f9b34fb len=5 withResp=true payload=[02 93 2C 4D 30]
? 2025-09-05 12:07:31.357 D/BluetoothConnManager: Adapter isConnected: true for Renpho-Scale (Status: CONNECTING)
+ 2025-09-05 12:07:31.359 I/BluetoothConnManager: Successfully connected to Renpho-Scale via adapter's isConnected flow.
? 2025-09-05 12:07:31.363 D/BluetoothConnManager: BluetoothEvent received: Connected(deviceName=Renpho-Scale, deviceAddress=FF:04:00:0D:1E:70) for Renpho-Scale
+ 2025-09-05 12:07:31.364 I/BluetoothConnManager: Event: Connected to Renpho-Scale (FF:04:00:0D:1E:70)
? 2025-09-05 12:07:31.365 D/BluetoothConnManager: BluetoothEvent received: DeviceMessage(message=Please step barefoot on the scale, deviceAddress=FF:04:00:0D:1E:70) for Renpho-Scale
? 2025-09-05 12:07:31.365 D/BluetoothConnManager: Event: Message from Renpho-Scale: Please step barefoot on the scale
? 2025-09-05 12:07:32.014 D/ScaleDeviceHandler: QN set weightScaleFactor=10.0 from opcode 0x12
? 2025-09-05 12:07:32.012 D/ScaleDeviceHandler: ← notify chr=0000fff1-0000-1000-8000-00805f9b34fb len=18 payload=[12 12 FF 70 1E 0D 00 04 FF 0F 07 0A 00 15 05 9F 30 CA]
? 2025-09-05 12:07:32.165 D/ScaleDeviceHandler: ← notify chr=0000fff1-0000-1000-8000-00805f9b34fb len=11 payload=[14 0B FF 00 00 01 00 00 00 00 1F]
? 2025-09-05 12:07:32.178 D/ScaleDeviceHandler: RAW notify: 140bff000001000000001f
? 2025-09-05 12:07:32.180 D/ScaleDeviceHandler: QN weight=0.0 kg, r1=0.0, r2=0.0 (scale=10.0)
+ 2025-09-05 12:08:04.414 I/BluetoothConnManager: disconnect() called for Renpho-Scale. Active communicator: true, Status: CONNECTED
? 2025-09-05 12:08:04.416 D/BluetoothConnManager: Adapter isConnected: false for Renpho-Scale (Status: DISCONNECTING)
[... cut ...]
? 2025-09-05 12:08:50.835 D/BluetoothConnManager: BluetoothEvent received: Disconnected(deviceAddress=FF:04:00:0D:1E:70, reason=SUCCESS) for Renpho-Scale
+ 2025-09-05 12:08:50.836 I/BluetoothConnManager: Event: Disconnected from FF:04:00:0D:1E:70. Reason: SUCCESS
? 2025-09-05 12:08:50.837 D/BluetoothConnManager: Disconnected event: Releasing active communicator: GattScaleAdapter
? 2025-09-05 12:08:50.838 D/BluetoothConnManager: Disconnected event: Active communicator released and set to null.

```

woaloo, 2025-09-05:
[image]

[image]

oliexdev, 2025-09-05:
could you do 
Create a User
Grant BT permissions
Add BT Scale
Connect to the scale
Step on scale 
Export log file and read on the scale the weight value and post it here

oliexdev, 2025-09-10:
Could be possible but very unlikely we have to wait if @keeperofdakeys  have some time to fix that

omersiar, 2025-11-29:
| Field | Value |
|---|---|
| Time | 2025-11-29 05:59:46.389 |
| Session ID | 19acdfb3f56 |
| App | openScale |
| Version | 3.0-debug (69) |
| Package | com.health.openscale.debug |
| Build Type | debug |
| Device | Xiaomi Redmi Note 8 |
| Android | 15 (API 35) |
| Build ID | lineage_ginkgo-userdebug 15 BP1A.250505.005 18182c761f |
| Build Time | 2025-11-16T14:18:22Z |
| Git SHA | 23c6423 |

```diff
? 2025-11-29 05:59:46.386 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.debug/files/logs/openScale_current_log.txt
? 2025-11-29 05:59:46.390 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.debug/files/logs/openScale_current_log.txt
+ 2025-11-29 05:59:46.391 I/openScaleLog: File logging enabled during runtime – started fresh session log.
? 2025-11-29 05:59:46.402 D/openScaleLog: Markdown diff block closed.
+ 2025-11-29 06:00:00.557 I/UserSettingsRepository: Saving device snapshot: addr=FF:04:00:10:93:AD, name=Renpho-Scale, uuids=1, hint=QN Scale
? 2025-11-29 06:00:08.644 D/BluetoothTopBar: User clicked bluetooth icon connect → trying to connect to saved device Renpho-Scale
+ 2025-11-29 06:00:08.647 I/BleConnector: Attempting to connect to Renpho-Scale
? 2025-11-29 06:00:08.648 D/BleConnector: Switching to new device: Releasing active communicator: null
? 2025-11-29 06:00:08.649 D/BleConnector: Switching to new device: Active communicator released and set to null.
? 2025-11-29 06:00:08.651 D/ScaleHandlerFactory: createCommunicator: Searching for communicator for 'Renpho-Scale' (FF:04:00:10:93:AD). Handler hint: 'QN Scale'
+ 2025-11-29 06:00:08.654 I/ScaleHandlerFactory: Modern handler 'QN Scale' supports 'Renpho-Scale'.
+ 2025-11-29 06:00:08.667 I/ScaleHandlerFactory: Modern communicator 'GattScaleAdapter' created for 'Renpho-Scale' with linkMode=CONNECT_GATT.
+ 2025-11-29 06:00:08.670 I/BleConnector: ActiveCommunicator successfully created: GattScaleAdapter. Starting observation job...
? 2025-11-29 06:00:08.671 D/BleConnector: Adapter isConnected: false for Renpho-Scale (Status: CONNECTING)
+ 2025-11-29 06:00:08.672 I/BleConnector: Adapter no longer reports connected for Renpho-Scale. Current status: CONNECTING. Expecting Disconnected Event.
+ 2025-11-29 06:00:13.603 I/GattScaleAdapter: Found FF:04:00:10:93:AD → stop scan + connect
? 2025-11-29 06:00:15.567 D/GattScaleAdapter: Services discovered for FF:04:00:10:93:AD
? 2025-11-29 06:00:15.569 D/QNHandler: attach()
? 2025-11-29 06:00:15.570 D/QNHandler: handleConnected(userId=1, height=189.0, age=18)
? 2025-11-29 06:00:15.573 D/QNHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe1-0000-1000-8000-00805f9b34fb → false
? 2025-11-29 06:00:15.574 D/GattScaleAdapter: → read from chr=00002a00-0000-1000-8000-00805f9b34fb svc=00001800-0000-1000-8000-00805f9b34fb
? 2025-11-29 06:00:15.574 D/QNHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe2-0000-1000-8000-00805f9b34fb → false
? 2025-11-29 06:00:15.576 D/QNHandler: QN seenProtocolType=0 unitByte=1
? 2025-11-29 06:00:15.579 D/QNHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe3-0000-1000-8000-00805f9b34fb → false
? 2025-11-29 06:00:15.581 D/QNHandler: hasCharacteristic: 0000ffe0-0000-1000-8000-00805f9b34fb/0000ffe4-0000-1000-8000-00805f9b34fb → false
? 2025-11-29 06:00:15.582 D/BleConnector: Adapter isConnected: true for Renpho-Scale (Status: CONNECTING)
+ 2025-11-29 06:00:15.584 I/BleConnector: Successfully connected to Renpho-Scale via adapter's isConnected flow.
? 2025-11-29 06:00:15.588 D/BleConnector: BluetoothEvent received: Connected(deviceName=Renpho-Scale, deviceAddress=FF:04:00:10:93:AD) for Renpho-Scale
+ 2025-11-29 06:00:15.589 I/BleConnector: Event: Connected to Renpho-Scale (FF:04:00:10:93:AD)
? 2025-11-29 06:00:15.590 D/BleConnector: BluetoothEvent received: DeviceMessage(message=Please step barefoot on the scale, deviceAddress=FF:04:00:10:93:AD) for Renpho-Scale
? 2025-11-29 06:00:15.591 D/BleConnector: Event: Message from Renpho-Scale: Please step barefoot on the scale
? 2025-11-29 06:00:16.176 D/GattScaleAdapter: ← received data chr=00002a00-0000-1000-8000-00805f9b34fb len=12 status=SUCCESS payload=[52 65 6E 70 68 6F 2D 53 63 61 6C 65]
? 2025-11-29 06:00:16.180 D/QNHandler: QN: notify chr=00002a00-0000-1000-8000-00805f9b34fb payload=[52 65 6E 70 68 6F 2D 53 63 61 6C 65] ascii=Renpho-Scale
? 2025-11-29 06:00:16.207 D/GattScaleAdapter: → read from chr=00002a29-0000-1000-8000-00805f9b34fb svc=0000180a-0000-1000-8000-00805f9b34fb
? 2025-11-29 06:00:16.235 D/GattScaleAdapter: ← received data chr=00002a29-0000-1000-8000-00805f9b34fb len=19 status=SUCCESS payload=[51 69 6E 67 20 4E 69 75 20 54 65 63 68 6E 6F 6C 6F 67 79]
? 2025-11-29 06:00:16.237 D/QNHandler: QN: notify chr=00002a29-0000-1000-8000-00805f9b34fb payload=[51 69 6E 67 20 4E 69 75 20 54 65 63 68 6E 6F 6C 6F 67 79] ascii=Qing Niu Technology
? 2025-11-29 06:00:16.262 D/GattScaleAdapter: → read from chr=00002a26-0000-1000-8000-00805f9b34fb svc=0000180a-0000-1000-8000-00805f9b34fb
? 2025-11-29 06:00:16.293 D/GattScaleAdapter: ← received data chr=00002a26-0000-1000-8000-00805f9b34fb len=5 status=SUCCESS payload=[56 31 30 2E 30]
? 2025-11-29 06:00:16.295 D/QNHandler: QN: notify chr=00002a26-0000-1000-8000-00805f9b34fb payload=[56 31 30 2E 30] ascii=V10.0
[... cut ...]
? 2025-11-29 06:00:17.192 D/GattScaleAdapter: → write to chr=0000fff2-0000-1000-8000-00805f9b34fb svc=0000fff0-0000-1000-8000-00805f9b34fb len=13 withResp=true payload=[A0 0D FF FE FF EE 01 1C 06 86 03 02 45]
? 2025-11-29 06:00:17.289 D/GattScaleAdapter: ← received data chr=0000fff1-0000-1000-8000-00805f9b34fb len=6 status=SUCCESS payload=[A1 06 FF FE 00 A4]
? 2025-11-29 06:00:17.292 D/QNHandler: QN: unhandled opcode=0xa1 payload=[A1 06 FF FE 00 A4]
? 2025-11-29 06:00:17.297 D/GattScaleAdapter: ← write response chr=0000fff2-0000-1000-8000-00805f9b34fb len=13 status=SUCCESS payload=[A0 0D FF FE FF EE 01 1C 06 86 03 02 45]
```