oliexdev/openScale #1167: Stone (st) & pound (lb) as a unit.

bsimmo, 2025-08-30:
Feature Request
Add Stone - pound as a unit, not "Stone.decimal fraction" 
I'm from UK so switch between stone-lb _and_ kg, but never use decimal stones

What would I like
I would like to see the option of displaying Stone & Pounds.
Currently it is displayed (from what I see) as a stone fraction e.g 13.7 stone.  There is not really such a thing as it is not metric, but imperial. There are 14 pounds (lb) to a stone (st).
So 87 kg = 12st 10lb

**Additional context**
UK uses Stone-pound and kg interchangeably.  Usually kg for science and doctor/medical or fitness training, but Stone & Pounds for general talk, and that varies with age/people.  So we flip between them as needed

Should mention
USA use just pounds for body weight (can anyone confirm)  
So they would say 192 pounds

How
In maths the conversion is
kg/0.45359237  (weight in pounds)
- get stones with floor division by 14
- get pounds with modulus 14

I hope that makes sense.

bsimmo, 2025-08-30:
> In the apps you already use, how do you typically enter stone–pounds—two
> separate inputs (stones / pounds) or a single field like “12 st 10 lb”? Or
> do you input everything in lb or st and only the displaying is xx st xx lb?
> In those apps, is the format the same for UK and US users, or do you
> usually see the US as pounds-only?
> Note: the app isn’t designed for frequent unit switching; I’m aiming for a
> per-profile default rather than a constant toggle.

Either as scrolling field from a list. (Not the best method)

Or
as two separate numbers fields.
One for stone, one from pounds.
type it in or arrows or drop downs

Or
Preformat fields for easy typing
 []st[]lb
Just type 1610 to get the numbers into the boxes []

Seems to vary, but oddly don't really notice.

It's a right pain.
Luckily we don't use ounce(oz) for body weight, as there are 16 of them in
a pound.

Hope that makes sense.
No rush, just a nice to have.

To make life fun, we would probably use kg for body parts, certainly g for
small parts.

It's like our distances on roads, miles and meters for many of us.

oliexdev, 2025-09-03:
try out the latest beta build on GitHub. You should enter in st and lb and the values should also be displayed as "xx st xx lb" if you have switched the measurement unit to ST.