oliexdev/openScale #1169: Tested WORKING setups with V3 beta

bsimmo, 2025-08-30:
I thought it might be useful for people to list actual working setups with the v3 beta.
Just so you know :-)

Build : Using 3.0 beta build 30 August 25 (68)

Scale : Silvercrest SBF77
Android Version : 15  (1st May 25 security)
Device : Moto Edge 50 Pro

Can get users, connect and get data.
Tried a few times now.

will check in more detail if it all looks correct.

oliexdev, 2025-09-02:
Thanks but could you test the [latest beta build ](https://github.com/oliexdev/openScale/releases/tag/beta-build) again? 
I’ve migrated the old Java scale handler to the modern Kotlin handler.

oliexdev, 2025-09-02:
Thanks for your feedback I have fixed your two hints in the latest build, lets see if you can connect to your scale.

oliexdev, 2025-09-07:
please post the log file without one I sticking in the dark.

closed 2025-10-18 by oliexdev

bsimmo, 2025-10-18:
| Field | Value |
|---|---|
| Time | 2025-10-18 13:03:54.566 |
| Session ID | 199f7347646 |
| App | openScale |
| Version | 3.0-beta (69) |
| Package | com.health.openscale.beta |
| Build Type | beta |
| Device | motorola motorola edge 50 pro |
| Android | 15 (API 35) |
| Build ID | V1UMS35H.10-67-7-2 |
| Build Time | 2025-10-18T10:25:37Z |
| Git SHA | c85de04 |

```diff
? 2025-10-18 13:03:54.566 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.beta/files/logs/openScale_current_log.txt
? 2025-10-18 13:03:54.567 D/openScaleLog: Initial markdown-diff headers written to: /storage/emulated/0/Android/data/com.health.openscale.beta/files/logs/openScale_current_log.txt
+ 2025-10-18 13:03:54.567 I/openScaleLog: File logging enabled during runtime – started fresh session log.
? 2025-10-18 13:04:02.402 D/BluetoothTopBar: User clicked bluetooth icon connect → trying to connect to saved device Debug
+ 2025-10-18 13:04:02.405 I/BleConnector: Attempting to connect to Debug
? 2025-10-18 13:04:02.406 D/BleConnector: Switching to new device: Releasing active communicator: null
? 2025-10-18 13:04:02.407 D/BleConnector: Switching to new device: Active communicator released and set to null.
? 2025-10-18 13:04:02.408 D/ScaleHandlerFactory: createCommunicator: Searching for communicator for 'Debug' (C7:34:11:85:8F:E0). Handler hint: 'Debug'
+ 2025-10-18 13:04:02.410 I/ScaleHandlerFactory: Modern handler 'Debug' supports 'Debug'.
+ 2025-10-18 13:04:02.415 I/ScaleHandlerFactory: Modern communicator 'GattScaleAdapter' created for 'Debug' with linkMode=CONNECT_GATT.
+ 2025-10-18 13:04:02.415 I/BleConnector: ActiveCommunicator successfully created: GattScaleAdapter. Starting observation job...
? 2025-10-18 13:04:02.417 D/BleConnector: Adapter isConnected: false for Debug (Status: CONNECTING)
+ 2025-10-18 13:04:02.418 I/BleConnector: Adapter no longer reports connected for Debug. Current status: CONNECTING. Expecting Disconnected Event.
+ 2025-10-18 13:04:10.353 I/GattScaleAdapter: Found C7:34:11:85:8F:E0 → stop scan + connect
? 2025-10-18 13:04:23.447 D/GattScaleAdapter: Services discovered for C7:34:11:85:8F:E0
? 2025-10-18 13:04:23.450 D/DebugGattHandler: attach()
? 2025-10-18 13:04:23.451 D/DebugGattHandler: handleConnected(userId=1, height=170.0, age=18)
? 2025-10-18 13:04:23.452 D/DebugGattHandler: Connected in Debug mode. Dumping full GATT services/characteristics…
? 2025-10-18 13:04:23.453 D/DebugGattHandler: === GATT Service Dump BEGIN ===
? 2025-10-18 13:04:23.454 D/DebugGattHandler: Service 0x1800
? 2025-10-18 13:04:23.455 D/DebugGattHandler:   └─ Char 0x2a00 props=READ
? 2025-10-18 13:04:23.456 D/DebugGattHandler:   └─ Char 0x2a01 props=READ
? 2025-10-18 13:04:23.457 D/DebugGattHandler:   └─ Char 0x2a02 props=READ|WRITE
? 2025-10-18 13:04:23.457 D/DebugGattHandler:   └─ Char 0x2a03 props=WRITE
? 2025-10-18 13:04:23.458 D/DebugGattHandler:   └─ Char 0x2a04 props=READ
? 2025-10-18 13:04:23.459 D/DebugGattHandler: Service 0x1801
? 2025-10-18 13:04:23.460 D/DebugGattHandler:   └─ Char 0x2a05 props=INDICATE
? 2025-10-18 13:04:23.461 D/DebugGattHandler: Service 0x180a
? 2025-10-18 13:04:23.462 D/DebugGattHandler:   └─ Char 0x2a24 props=READ
? 2025-10-18 13:04:23.463 D/DebugGattHandler:   └─ Char 0x2a26 props=READ
? 2025-10-18 13:04:23.464 D/DebugGattHandler:   └─ Char 0x2a27 props=READ
? 2025-10-18 13:04:23.465 D/DebugGattHandler:   └─ Char 0x2a28 props=READ
? 2025-10-18 13:04:23.466 D/DebugGattHandler:   └─ Char 0x2a29 props=READ
? 2025-10-18 13:04:23.466 D/DebugGattHandler: Service 0x180f
? 2025-10-18 13:04:23.466 D/DebugGattHandler:   └─ Char 0x2a2a props=READ
? 2025-10-18 13:04:23.467 D/DebugGattHandler:   └─ Char 0x2a19 props=READ
? 2025-10-18 13:04:23.467 D/DebugGattHandler: Service 0x1805
? 2025-10-18 13:04:23.467 D/DebugGattHandler:   └─ Char 0x2a2b props=READ|WRITE|NOTIFY
? 2025-10-18 13:04:23.468 D/DebugGattHandler: Service 0x181c
? 2025-10-18 13:04:23.468 D/DebugGattHandler:   └─ Char 0x2a85 props=READ|WRITE
? 2025-10-18 13:04:23.468 D/DebugGattHandler:   └─ Char 0x2a8c props=READ|WRITE
? 2025-10-18 13:04:23.469 D/DebugGattHandler:   └─ Char 0x2a99 props=READ|WRITE|NOTIFY
? 2025-10-18 13:04:23.469 D/DebugGattHandler:   └─ Char 0x2a9a props=READ
? 2025-10-18 13:04:23.470 D/DebugGattHandler:   └─ Char 0x2a9f props=WRITE|INDICATE
? 2025-10-18 13:04:23.470 D/DebugGattHandler: Service 0x181d
? 2025-10-18 13:04:23.470 D/DebugGattHandler:   └─ Char 0x2a9e props=READ
? 2025-10-18 13:04:23.470 D/DebugGattHandler:   └─ Char 0x2a9d props=INDICATE
? 2025-10-18 13:04:23.471 D/DebugGattHandler: Service 0x181b
? 2025-10-18 13:04:23.471 D/DebugGattHandler:   └─ Char 0x2a9b props=READ
? 2025-10-18 13:04:23.471 D/DebugGattHandler:   └─ Char 0x2a9c props=INDICATE
? 2025-10-18 13:04:23.471 D/DebugGattHandler: Service 0xffff
? 2025-10-18 13:04:23.472 D/DebugGattHandler:   └─ Char 0x0000 props=READ|WRITE
? 2025-10-18 13:04:23.469 D/DebugGattHandler:   └─ Char 0x2a8e props=READ|WRITE
? 2025-10-18 13:04:23.472 D/DebugGattHandler:   └─ Char 0x0001 props=READ|WRITE|NOTIFY
? 2025-10-18 13:04:23.472 D/DebugGattHandler:   └─ Char 0x0002 props=READ|WRITE
[... 30 lines cut ...]
? 2025-10-18 13:04:24.432 D/GattScaleAdapter: ← received data chr=00002a26-0000-1000-8000-00805f9b34fb len=3 status=SUCCESS payload=[37 2E 38]
? 2025-10-18 13:04:24.433 D/DebugGattHandler: NOTIFY chr=0x2a26  payload=[37 2E 38]  ascii=7.8
[... cut ...]
? 2025-10-18 13:05:00.456 D/DebugGattHandler: detach()
? 2025-10-18 13:05:00.456 D/BleConnector: BluetoothEvent received: Disconnected(deviceAddress=C7:34:11:85:8F:E0, reason=SUCCESS) for Debug
+ 2025-10-18 13:05:00.457 I/BleConnector: Event: Disconnected from C7:34:11:85:8F:E0. Reason: SUCCESS
? 2025-10-18 13:05:00.458 D/BleConnector: Disconnected event: Releasing active communicator: GattScaleAdapter
? 2025-10-18 13:05:00.458 D/BleConnector: Disconnected event: Active communicator released and set to null.

```
With today's new build.

bsimmo, 2025-10-18:
I enabled the Bluetooth as debug, enabled the log file creation in setting, saved it. Then copy pasted it here. Twice now. Which you nicely formatted to make it easy to read.
Which is what the V3 beta invite thread says iirc.

oliexdev, 2025-10-18:
because you are using the DEBUG Bluetooth Handler which just dump the GATT Service.

bsimmo, 2025-10-18:
You warning says my measurements (weight etc aka privacy and not saved to the log).
I'm confused what you want..
Or what you debug mode it then.

Anyway: You now have logs for debug enabled on BT device and debug not enabled on BT device.

I can't find any other logs.