oliexdev/openScale #1170: Measurement ID starts at 0 after reinstall & db import

fallingcats, 2025-09-05:
**Describe the bug**
I'm not sure whether it's a bug, but it seems very unusual to me that openscale db indices aren't strictly monotonically increasing

**To Reproduce**
Steps to reproduce the behavior:

0. (Delete first couple of measurements from the db?)
1. Reinstall app
2. Import backup
3. Set up `openScale-sync` with mqtt
4. Notice that new measurements start at id=0

Reproduced with [latest dev version](https://github.com/oliexdev/openScale/releases/tag/dev-build): No? The newest one isn't tagged as dev for some reason https://github.com/oliexdev/openScale/commit/3c323aee0c8ec12c72ae85e37996676550c59fd0

**Expected behavior**
DB indices should only ever increase

**Additional context**
I've implemented sorting by ID in openscale sync (mqtt) recently, and this kinda throws a wrench into things.

fallingcats, 2025-09-05:
I also want to add that I think this wasn't always the case

oliexdev, 2025-09-05:
no that's normal don't depend on the id while sorting use the timestamp. The id is auto generated and handled by the room database. It is just a unique identifier for each measurment

closed 2025-09-05 by oliexdev