oliexdev/openScale #1172: SBF 73 is already supported (stable + beta)

diggit, 2025-09-07:
Hi,
I just want to let you know the SBF 73 (from Lidl) works with current stable (f-droid) build and also works with beta 3.0 (68) apk.

My observations:

Stable 2.x.x:
- has a same caveat as SBF 70 regarding the user naming

Beta 3.0 (68):
- identified by app as "Sanitas SBF **72**"

I can test stuff and collect logs, in case you need that.

This is just a report. Feel free to close it if you don't need to test/collect anything with this scale.

oliexdev, 2025-09-07:
Thanks for your feedback, did you test the latest beta verison on GitHub because there I migrated all bluetooth handlers to kotlin.

diggit, 2025-09-07:
> did you test the latest beta verison on GitHub because there I migrated all bluetooth handlers to kotlin.

Yes, taken from releases, commit c88c388fe32f982487338f22c385ea2b838ebb92

I played with beta a little bit a I was able to trigger app crash, when it was connecting/communicating with scale. It looked like if user name is not same as in the scale, it app will crash instead of no getting advanced metrics (2.x.x behavior).

oliexdev, 2025-09-07:
please post the log file

diggit, 2025-09-07:
No crash, but also no data are read out.

3.x log:  [openScale_connected_no_data.txt](https://github.com/user-attachments/files/22197998/openScale_connected_no_data.txt)

2.x log (for comparison): [openScale_2025-09-07_20-11.txt](https://github.com/user-attachments/files/22198001/openScale_2025-09-07_20-11.txt)

even 2.x took some time before new measurements appeared.

oliexdev, 2025-09-27:
@diggit SBF 72 / 73 should be working now with the latest beta build

closed 2025-09-28 by diggit